JorenSix/Olaf #4: Audio not  matching

internetztube, 2020-11-28:
Hello Joren!
First of all, I want to thank you for publishing this project on Github!

Have the problem that an audio snippet is not recognized.

```
olaf store store.mp3
```
```
olaf query external-stream.mp3
```

Side note, in my case the monitor command throws an error, maybe this is related.
```
olaf monitor external-stream.mp3
```
```
$ olaf monitor external-stream.mp3
1/1_0s external-stream.mp3 matches trimmed.mov q_to_ref_time_delta: 0.00, q_time: 5.44, score: 146, match_id: 779030374, ref_start: 0.29, ref_stop: 5.44
1/1_0s external-stream.mp3 Proccessed 471 fp's from 7.0s in 0.004s (1962 times realtime)
1/1_0s external-stream.mp3 Audio file 151390292 not found or unreadable.
1/1_7s external-stream.mp3 Proccessed 502 fp's from 7.0s in 0.004s (1969 times realtime)
1/1_7s external-stream.mp3 Audio file 151390292 not found or unreadable.
1/1_14s external-stream.mp3 Proccessed 515 fp's from 7.0s in 0.003s (2020 times realtime)
1/1_14s external-stream.mp3 Audio file 151390292 not found or unreadable.
Traceback (most recent call last):
	9: from /usr/local/bin/olaf:460:in `<main>'
	8: from /usr/local/bin/olaf:460:in `each_with_index'
	7: from /usr/local/bin/olaf:460:in `each'
	6: from /usr/local/bin/olaf:461:in `block in <main>'
	5: from /usr/local/bin/olaf:84:in `monitor'
	4: from /usr/local/bin/olaf:189:in `with_converted_audio_part'
	3: from /usr/local/bin/olaf:88:in `block in monitor'
	2: from /usr/local/bin/olaf:88:in `each'
	1: from /usr/local/bin/olaf:95:in `block (2 levels) in monitor'
/usr/local/bin/olaf:95:in `basename': no implicit conversion of nil into String (TypeError)
```

Maybe you can help me with that.

Greetings Fred
[Archive.zip](https://github.com/JorenSix/Olaf/files/5610462/Archive.zip)

JorenSix, 2020-12-01:
Thanks for the bug report and for providing test files. Much appreciated.

I was in the process of reshuffling a couple of things. Now the output is somewhat better documented and I removed a bit of stale code (an inefficient matcher). These changes had some unintended consequences.

Also fixed a bug where empty database results could return a match that was interpreted incorrectly (and result in the filename bug).

```
olaf query external-stream.mp3
query index,total queries, query name, match name, match id, match count (#), q to ref time delta (s), ref start (s), ref stop (s), query time (s)
1,1,external-stream.mp3,store.mp3 3812745049, 15, 0.09, 0.01, 0.02, 0.12
1,1,external-stream.mp3,store.mp3 3812745049, 12, 0.04, 0.00, 0.02, 0.07
Proccessed 13202 fp's from 153.2s in 0.095s (1617 times realtime)

olaf monitor external-stream.mp3
query index,total queries, query name, match name, match id, match count (#), q to ref time delta (s), ref start (s), ref stop (s), query time (s)
1/1_112s external-stream.mp3 matches store.mp3 3812745049, 10, -0.02, 0.02, 0.02, 0.00
```

closed 2020-12-01 by JorenSix

internetztube, 2020-12-01:
Many thanks, Joren! 🎉

internetztube, 2020-12-01:
👋 Joren!
The example files that I provided now work. But another example is now not working. Thanks in advance! 
Regards Fred
[falco_wiener_blut.zip](https://github.com/JorenSix/Olaf/files/5625682/falco_wiener_blut.zip)

JorenSix, 2020-12-03:
That example also works now (but barely) by rounding the frequency part of the hash instead of truncating it. I have better experience with longer queries...

To get better insights into Olafs retrieval performance I have included an evaluation script which can be used (or modified) for your dataset. It needs a folder with audio items and creates and launches queries. 

This is only a fist part: the analysis part is still in the works. With that info I will be able to optimise parameter set / fingerprint hash construction.

JorenSix, 2020-12-09:
Shuffled things around a bit for a better sensitivity while keeping a perfect specificity calculated using the evaluation script. 
This changed fundamental aspects of the algorithm and should improve matching. Please give it an other try.

internetztube, 2020-12-09:
Will give it a try later this week. Thanks for your effort!