JorenSix/Olaf #8: Segmentation fault on some WAV files (even after FFMPEG converts them to RAW)

konstantin-sancom, 2021-11-22:
`OS: Debian 10 x64`
`gcc version 8.3.0 (Debian 8.3.0-6)`

For some reason `olaf_c` goes to `segmentation fault` on some files.
It is simply reprodusable on `store` and on `query` mode.

`query` mode make output as:
`...`
`start key: 4365488144 stop key: 4365488154  results 0`
`start key: 4365488144 stop key: 4365488154  results 0`
`Segmentation fault`

`GDB` does not show paticular line, but shows :

`Program received signal SIGSEGV, Segmentation fault.`
`0x0000555555565003 in olaf_reader_read ()`
`(gdb) backtrace`
`#0  0x0000555555565003 in olaf_reader_read ()`
`#1  0x000055555556542d in olaf_stream_processor_process ()`
`#2  0x000055555555674e in main ()`
`(gdb)`

kvishnivetsky, 2022-08-29:
Hi!
Is there any update on this issue?

konstantin-sancom, 2023-01-13:
@JorenSix , this issue is still actual. Please, give us an advice, how to solve this?

JorenSix, 2023-01-13:
Hi could you send me a wav file where this happens?

konstantin-sancom, 2023-01-17:
Hi, @JorenSix . We checked it again - now it is allright.
The probem was in input RAW file format this time.