JorenSix/Olaf #15: Understanding the Output

apb2019, 2022-03-15:
For testing I have "stored" an mp3 (compilation.mp3) made from several tracks against which I have run 3 queries.
Query 1 = .mp3 of one of the tracks (made from the original .flac used to make the compilation) so would expect a single match but got the following:
1/3 TestAll54sec.mp3 match id, match count (#), q to ref time delta (s), ref start (s), ref stop (s), query time (s)
1, 3, TestAll54sec.mp3, 2146557281, /home/apb/Olaf-master/Compilation.mp3, 55, -2594.35, 2637.10, 2648.18, 53.83
1, 3, TestAll54sec.mp3, 2146557281, /home/apb/Olaf-master/Compilation.mp3, 6, -2594.34, 2643.14, 2643.14, 48.80

Query 2 = a 15sec clip from the same track as query 1
2, 3, Test15sec.mp3, 2146557281, /home/apb/Olaf-master/Compilation.mp3, 332, -2608.95, 2609.14, 2623.73, 14.78

Query 3 = the original .flac file used to make the .mp3 for query 1
3, 3, TestAll54secFLAC.flac, 2146557281, /home/apb/Olaf-master/Compilation.mp3, 84, -2594.35, 2637.10, 2648.03, 53.68

Questions:
1. What do match count (#), q to ref time delta (s), ref start (s), ref stop (s), query time (s) mean? other than -2594.35 which is the negative of the start time of the track in the compilation and 53.83 which seems to match the track length I'm not sure how to interpret the other numbers.
2. Any thoughts as to why query 1 would return 2 matches?
3. The only difference between this query and query 1 was the file used was the original .flac not an. mp3 made from the same file so shouldn't the results be the same?

diamondsw, 2022-03-25:
The output that comes currently seems like debug output, not useful output. I don't need to see floods of (start key:|Found key:|No Next Dup for key), I don't need to see internal database IDs. What I need to see is which reference files in the database this matches (with a single entry per item summarizing how much it matched), ranked by likelihood. I'll end up coding a perl wrapper script to do this, but it's a shame the built-in output isn't framed this way.

JorenSix, 2023-01-25:
I have improved the output a while back.
Indeed this was not meant to be checked in as such.

closed 2023-01-25 by JorenSix