JorenSix/Olaf #21: Possible BUG: There is no boundaies control in olaf_fp_extractor.c

konstantin-sancom, 2023-01-19:
Hi, @JorenSix 

There is no boundaries control for `fp_extractor->fingerprints.fingerprints`.
So, `[fp_extractor->fingerprints.fingerprintIndex` can be incremented and go out of initial allocated for `fp_extractor->fingerprints.fingerprints` memmory block. I causes memory corruption in other modules and unpredictable behaivior.
I fixed this with additional condition in `for` loops:
`for(int k = j+1; k < eventPoints->eventPointIndex && fp_extractor->fingerprints.fingerprintIndex < fp_extractor->config->maxFingerprints; k++){`
and
`for(int j = i+1; j < eventPoints->eventPointIndex && fp_extractor->fingerprints.fingerprintIndex < fp_extractor->config->maxFingerprints; j++){`

But I do not know how would this affect fingerprinting algotithmus.

konstantin-sancom, 2023-01-25:
Hi!
I have an update. There definitely is a problem whith array bounds control.
I got rid of it by tuning configuration parameters, but it is a BUG, really.
In my local code copy I added WARNINGS for those cases.

JorenSix, 2023-01-25:
Hi, thanks for the heads up. 

I will have a look at it once i have a bit more time :)

JorenSix, 2023-01-25:
I have just added a CI script which checks the following for each pushed version: does the code build (make and make mem), does the install script install the software, does the basic functionality work? 

The basic functionality  tested is 
* Storing files in the reference database
* Deleting a file from the reference database
* Querying the database and veryfing the results

This works on my local machine (compiled with clang) but gives a segmentation fault on ubuntu. The CI shows its benefits since now I can see if it is related to this bug or another bug...

JorenSix, 2023-01-25:
I have checked in a version with arrays bound checking. Indeed there was no check and it overflowed for many mp3 files (20%). The CI did its job and now finished completely so I think this bug is fixed. 

Strange that  macOS was not complaining about incorrect memory use which was hiding the problem a bit.

Thanks for bringing this to my attention.

closed 2023-01-25 by JorenSix