JorenSix/Olaf #26: Automated test failures

liscio, 2023-05-10:
I'm on an M1 Mac running Ventura. `gcc` is symlinked to `clang` on my system, which isn't unusual on macOS.

When I build the test binary using `make test`, and then run `bin/olaf_tests`, an assertion fails: 

`(array_size == 512), function olaf_max_filter, file olaf_max_filter_perceptual.c, line 43.`

Further, when I run `eval/olaf_functional_tests.rb` I get the following output:

```
PASS: Test dataset check: If not found, download it first! call e.g. 'ruby eval/olaf_download_dataset.rb'
1/1 1051039.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/1051039.mp3'
1/1 1071559.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/1071559.mp3'
1/1 1075784.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/1075784.mp3'
1/1 11266.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/11266.mp3'
1/1 147199.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/147199.mp3'
1/1 173050.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/173050.mp3'
1/1 189211.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/189211.mp3'
1/1 297888.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/297888.mp3'
1/1 612409.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/612409.mp3'
1/1 852601.mp3 SKIP: already stored audio
PASS: Command : olaf store 'dataset/ref/852601.mp3'
FAIL: Expected 10 stored, was 11
FAIL: Expected 10 stored, was 11
```

Related to: https://github.com/openjournals/joss-reviews/issues/5459

JorenSix, 2023-05-12:
Thanks for taking the time to try it out and test it.  Indeed the documentation for running the tests is lacking which gives room for confusion. 

The @make test@ runs a couple of (outdated) sanity checks which were used during development. I will replace it with the functional tests (which include the unit tests in a way) and make sure it runs fine. It seems the error mainly has to do with not clearing the database properly before running tests.

JorenSix, 2023-05-12:
The problem had indeed to do with a non-empty database when starting the functional tests (e.g. when a previous test is aborted). Now there is a warning if a non-empty database is detected. 

The database is not thrashed automatically since it can contain a large amounts of user data.

closed 2023-05-12 by JorenSix