JorenSix/Olaf #34: Understanding the Output from ESP32 Example

JonRLG, 2023-06-16:
Im currently trying to run the ESP32 example.

Unfortunately im struggling to understand the Output. It does look like its not working properly but I dont know what the Problem is.

Here is the Output from the Arduino IDE:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
[OLAF] Free memory before init: 273124 bytes
No home directory found, will use './db' as db folderMem DB for '852601.wav', 193.761s duration, 13207 fingerprints, 666 identifier
[OLAF] Free after init: 150444 bytes
[OLAF] Initialized with 32091 audio samples. Sample sample: 0.036510
Event point index: 0 
[OLAF] processed audio block index: 0
Event point index: 0 
[OLAF] processed audio block index: 1
Event point index: 0 
[OLAF] processed audio block index: 2
Event point index: 0 
[OLAF] processed audio block index: 3
Event point index: 0 
[OLAF] processed audio block index: 4
Event point index: 0 
[OLAF] processed audio block index: 5
Event point index: 0 
[OLAF] processed audio block index: 6
Event point index: 0 
[OLAF] processed audio block index: 7
Event point index: 0 
[OLAF] processed audio block index: 8
Event point index: 0 
[OLAF] processed audio block index: 9
Event point index: 0 
[OLAF] processed audio block index: 10
Event point index: 0 
[OLAF] processed audio block index: 11
Event point index: 0 
[OLAF] processed audio block index: 12
Event point index: 0 
...

...
[OLAF] processed audio block index: 106
Event point index: 29 
[OLAF] processed audio block index: 107
Event point index: 30 
[OLAF] processed audio block index: 108
Event point index: 30 
[OLAF] processed audio block index: 109
Event point index: 31 
FP index: 6 
0, 0.00, 0.00, , 0, 0.00, 0.00
[OLAF] processed audio block index: 110
Event point index: 23 
[OLAF] processed audio block index: 111
```

JorenSix, 2023-06-20:
Hi, the output is currently debug output which makes sure that audio blocks are flowing through the system. This seems to be the case.

But many things can go wrong. To make it work you need to sample (16kHz) and step through the audio according to the expected block size and step size (1024 and 128 samples). Some digital microphones have problems with a 16kHz sample rate. Make sure this is in order.

The first thing you might want to check is the "olaf mem" program and try it out on a computer for your recognition task. This is the same code which runs on the ESP 32. Only if this works correctly I would go over to the much harder to debug ESP32.

JonRLG, 2023-06-30:
Hi, this is the output from the example "no_mic_test". Im not using any microphone at the moment. 
Do you have any idea what could cause the example to not work?

JorenSix, 2023-07-24:
Hi the no mic test has two header files with data. The first header contains a list of target fingerprints for the audio you want to match. 

The second header contains audio data. Since there is no microphone or audio input, the 'audio' is read from the header file and fed to Olaf. When the end of the audio data is reached, the same audio is repeated over and over.

Unfortunately the RAM of the ESP32 is not that large and audio takes up quite a bit (byte) of space. This means that the audio in the header file is short. For fingerprint matching, Olaf needs a few seconds of audio. The audio in the header file is too short for a match. The repetition does not help for the match.  

The no_mic_test is essentially only checking if the audio can flow through the system and the matching step can be executed (but no matches are reported). The output is basically debug output.

To fix it you either need more audio (a larger header file, more RAM) or an actual microphone input.

JorenSix, 2023-09-25:
Hi, I have included a new working example with a INMP441 MEMS microphone and an ESP32. There is also a readme for the ESP32 folder with additional details. Please have a look.

JorenSix, 2023-09-25:
More details here: https://github.com/JorenSix/Olaf/tree/master/ESP32