JorenSix/Olaf #35: Missing square brackets around references.

ebezzam, 2023-06-21:
https://github.com/JorenSix/Olaf/blob/0173ff18411294ea0577569e6cbd52dbad8fd85d/paper.md?plain=1#L40

JorenSix, 2023-06-21:
There is a deliberate formatting difference when using square brackets or not. In some cases the references are meant to fit in the text. 

For example "a paper by Fenet et al (2020)" is without brackets while "there are a number of papers on blablabla [Fenet 2020; Wang 2003]" is with brackets.

closed 2023-06-21 by JorenSix

ebezzam, 2023-06-22:
Got it, could you still format the citation for "Cortes et. al, 2022" entry so that it's like the others? The long list of names on lines 41-42 and 51-52 is a but awkward.

JorenSix, 2023-06-23:
Fixed, the problem was the use of `authors = {{author list}}`. With this formatting the author list is used literally, when using `authors = {author list}` it is correctly replaced by an 'et al'.