JorenSix/Olaf #43: Missing file

sonictruth, 2023-08-08:
Hi,

I'm trying to compile the web version but 'wasm/js/olaf_processor_edit.js' is missing. 
And I'm getting this error in Chrome:
**_DOMException: Failed to construct 'AudioWorkletNode': AudioWorkletNode cannot be created: The node name 'olaf-processor' is not defined in AudioWorkletGlobalScope._**

Can you please upload this file also to the repository ?

Thank you!

zheffie, 2023-09-04:
Same issue here. `make web` fails due to missing file `wasm/js/olaf_processor_edit.js`

JorenSix, 2023-09-27:
The issue has been fixed, I have shifted around a couple of files and the `make web` now works as expected and compiles the right files. The Olaf WASM demo now also includes an audio resampling library to make sure that the audio is in the format Olaf expects. Check out the online demo here:

https://0110.be/posts/Acoustic_fingerprinting_in_the_browser_with_Olaf