JorenSix/Olaf #46: recognize short audio signal with distinct tones

zheffie, 2023-11-19:
[beeping.zip](https://github.com/JorenSix/Olaf/files/13401773/beeping.zip)
I'm having a hard time recognizing my audio signal consistently. It is a short sequence of well defined tones (audio file in attachment).
Could you recommend `olaf_config.c` configuration to fingerprint and recognize this signal using ESP32?
Is it possible to recognize and distinguis between multiple audio signals on the ESP? 
thanks in advance

JorenSix, 2023-12-06:
Hi,

Olaf is indeed not fit for this type of audio. The ideal fit for Olaf is longer duration - over five seconds - music which contains many different notes. For tone patterns I would suggest to look for efficient [generalized goertzel implementations](
https://0110.be/posts/Is_a_frequency_present_in_a_signal%3F_A_C_implementation._).

Once you can detect a frequency you can also detect for how long this frequency was present and compare that with a template duration or pattern.

Regards

Joren

closed 2023-12-06 by JorenSix

zheffie, 2023-12-06:
I indeed already ventured in the Goertzel implementations. Thanks for confirming.