JorenSix/Olaf #50: AltiVec code fails to compile on ppc

barracuda156, 2024-05-12:
@JorenSix Could you please take a look? AltiVec code does not compile on `ppc`:
```
--->  Building olaf-devel
Executing:  cd "/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_audio_olaf-devel/olaf-devel/work/Olaf-4442e91b65d151211afc22a161c30fb07cf6ce3b" && /usr/bin/make -j6 -w compile 
make: Entering directory `/opt/local/var/macports/build/_opt_PPCSnowLeopardPorts_audio_olaf-devel/olaf-devel/work/Olaf-4442e91b65d151211afc22a161c30fb07cf6ce3b'
/opt/local/bin/gcc-mp-13 -c src/pffft.c 					-W -Wall -std=gnu11 -pedantic -O2 #pfft needs M_PI and other constants not in the ANSI c standard
src/pffft.c: In function 'ld_ps1':
src/pffft.c:107:45: warning: implicit declaration of function 'vec_lde' [-Wimplicit-function-declaration]
  107 | inline v4sf ld_ps1(const float *p) { v4sf v=vec_lde(0,p); return vec_splat(vec_perm(v, v, vec_lvsl(0, p)), 0); }
      |                                             ^~~~~~~
src/pffft.c:107:45: error: incompatible types when initializing type 'v4sf' {aka '__vector(4) float'} using type 'int'
src/pffft.c:107:66: warning: implicit declaration of function 'vec_splat' [-Wimplicit-function-declaration]
  107 | inline v4sf ld_ps1(const float *p) { v4sf v=vec_lde(0,p); return vec_splat(vec_perm(v, v, vec_lvsl(0, p)), 0); }
      |                                                                  ^~~~~~~~~
src/pffft.c:107:76: warning: implicit declaration of function 'vec_perm' [-Wimplicit-function-declaration]
  107 | inline v4sf ld_ps1(const float *p) { v4sf v=vec_lde(0,p); return vec_splat(vec_perm(v, v, vec_lvsl(0, p)), 0); }
      |                                                                            ^~~~~~~~
src/pffft.c:107:91: warning: implicit declaration of function 'vec_lvsl' [-Wimplicit-function-declaration]
  107 | inline v4sf ld_ps1(const float *p) { v4sf v=vec_lde(0,p); return vec_splat(vec_perm(v, v, vec_lvsl(0, p)), 0); }
      |                                                                                           ^~~~~~~~
src/pffft.c:107:1: error: AltiVec argument passed to unprototyped function
  107 | inline v4sf ld_ps1(const float *p) { v4sf v=vec_lde(0,p); return vec_splat(vec_perm(v, v, vec_lvsl(0, p)), 0); }
      | ^~~~~~
src/pffft.c: In function 'validate_pffft_simd':
src/pffft.c:102:35: warning: implicit declaration of function 'vec_splat_u8' [-Wimplicit-function-declaration]
  102 | #  define VZERO() ((vector float) vec_splat_u8(0))
      |                                   ^~~~~~~~~~~~
src/pffft.c:221:25: note: in expansion of macro 'VZERO'
  221 |   t = a0; u = a1; t.v = VZERO();
      |                         ^~~~~
src/pffft.c:221:3: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
  221 |   t = a0; u = a1; t.v = VZERO();
      |   ^
src/pffft.c:104:21: warning: implicit declaration of function 'vec_add' [-Wimplicit-function-declaration]
  104 | #  define VADD(a,b) vec_add(a,b)
      |                     ^~~~~~~
src/pffft.c:223:9: note: in expansion of macro 'VADD'
  223 |   t.v = VADD(a1.v, a2.v);
      |         ^~~~
src/pffft.c:223:3: error: AltiVec argument passed to unprototyped function
  223 |   t.v = VADD(a1.v, a2.v);
      |   ^
src/pffft.c:103:21: warning: implicit declaration of function 'vec_madd' [-Wimplicit-function-declaration]
  103 | #  define VMUL(a,b) vec_madd(a,b, VZERO())
      |                     ^~~~~~~~
src/pffft.c:225:9: note: in expansion of macro 'VMUL'
  225 |   t.v = VMUL(a1.v, a2.v);
      |         ^~~~
src/pffft.c:225:3: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
  225 |   t.v = VMUL(a1.v, a2.v);
      |   ^
src/pffft.c:225:3: error: AltiVec argument passed to unprototyped function
src/pffft.c:227:3: error: AltiVec argument passed to unprototyped function
  227 |   t.v = VMADD(a1.v, a2.v,a0.v);
      |   ^
src/pffft.c:109:60: warning: implicit declaration of function 'vec_mergeh' [-Wimplicit-function-declaration]
  109 | #  define INTERLEAVE2(in1, in2, out1, out2) { v4sf tmp__ = vec_mergeh(in1, in2); out2 = vec_mergel(in1, in2); out1 = tmp__; }
      |                                                            ^~~~~~~~~~
src/pffft.c:230:3: note: in expansion of macro 'INTERLEAVE2'
  230 |   INTERLEAVE2(a1.v,a2.v,t.v,u.v);
      |   ^~~~~~~~~~~
src/pffft.c:230:3: error: AltiVec argument passed to unprototyped function
src/pffft.c:109:89: warning: implicit declaration of function 'vec_mergel' [-Wimplicit-function-declaration]
  109 | #  define INTERLEAVE2(in1, in2, out1, out2) { v4sf tmp__ = vec_mergeh(in1, in2); out2 = vec_mergel(in1, in2); out1 = tmp__; }
      |                                                                                         ^~~~~~~~~~
src/pffft.c:230:3: note: in expansion of macro 'INTERLEAVE2'
  230 |   INTERLEAVE2(a1.v,a2.v,t.v,u.v);
      |   ^~~~~~~~~~~
src/pffft.c:230:3: error: AltiVec argument passed to unprototyped function
src/pffft.c:111:60: warning: left-hand operand of comma expression has no effect [-Wunused-value]
[... 594 lines cut ...]
src/pffft.c:437:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:437:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
src/pffft.c:437:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:437:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:438:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
  438 |       ci3 = VADD(cc_ref(i  , 1), VADD(SVMUL(tr12, ti2),SVMUL(tr11, ti3)));
      |       ^~~
src/pffft.c:438:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:438:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
src/pffft.c:438:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:438:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:439:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
  439 |       cr5 = VADD(SVMUL(ti11, tr5), SVMUL(ti12, tr4));
      |       ^~~
src/pffft.c:439:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:439:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
src/pffft.c:439:7: error: AltiVec argument passed to unprototyped function
src/pffft.c:440:7: error: cannot convert a value of type 'int' to vector type '__vector(4) float' which has different size
  440 |       ci5 = VADD(SVMUL(ti11, ti5), SVMUL(ti12, ti4));
      |       ^~~
. . .
```
Does it require VSX or some other issue causes the error?

(If AltiVec code cannot be fixed, perhaps an easy way would be to disable it, falling back to scalar mode.)

JorenSix, 2024-05-15:
Hi,

PPC, vintage! Anyway, Olaf should work with a bit of poking. I think you can skip the AltiVec code by providing a flag `PFFFT_SIMD_DISABLE` to the compile step of pffft.c  See: https://bitbucket.org/jpommier/pffft/src/master/pffft.c

Something like `gcc -c src/pffft.c -DPFFFT_SIMD_DISABLE -W -Wall -std=gnu11 -pedantic -O2`

barracuda156, 2024-05-15:
@JorenSix Disabling has been done in https://github.com/macports/macports-ports/commit/c910209763bb0e69641c48e0cc6798d4876579f5

Is it fixable properly though? I can test it on my end.

JorenSix, 2024-05-16:
Now it should be fine with the same patch as done by MacPorts. I have not tested it on PPC. I do not know if there is a better way to use perhaps other speed-ups available on older PPC machines. These optimisations should be done in the upstream pffft package:  https://bitbucket.org/jpommier/pffft/src/master/

Thanks for letting me know about this issue and making Olaf better!

barracuda156, 2024-05-16:
@JorenSix Just for the sake of clarity, Apple systems (aside of very old G3 machines) do support AltiVec; what they do not support is post-ISA 2.03 additions like VSX.

JorenSix, 2024-05-16:
Thanks for the additional info. I have notified the maintainer of pffft and I hope it will get some attention upstream.

closed 2024-05-16 by JorenSix

barracuda156, 2024-05-16:
@JorenSix Thank you!

JorenSix, 2024-05-17:
@barracuda156 I contacted the pffft maintainer and there was a worry about selectively disabling altivec. If it is not done right, it might end up disabling the speedup for processors which actually do support it. So pffft might let users disable speed-ups that do not work (opt-out). 

For Olaf the reverse holds: it should 'just work' if you want to get the most out of your system you might want to experiment with enabling some speed-ups (opt-in).

Unless there is a known, well documented list of PPC processors/ with associated C-flags which map perfectly to the several speed-ups this seems a reasonable approach.

barracuda156, 2024-05-17:
AltiVec as such should be fully supported on G4 and G5, which roughly correspond to ISA 2.03. Since this code fails to compile on G5, I suspect that either the code uses non-AltiVec extensions of the later ISA, like VSX, or it has a bug of some sort preventing it from compiling.
I cannot verify if it builds on ISA-2.06-supporting CPUs, since I have no hardware of that kind.

Unless the code is fixed, it is safe to disable it for Apple, since the latest cpu which macOS can run on is G5. And obviously a slowly working software is better than a broken one.

Situation will be more complicated on *BSD, which may run on a wider range of PowerPC hardware.
@pkubaj Maybe you could comment on this?